opendata-stuttgart/airrohr-firmware-flasher #1: Settings erasure option

Informatic, 2018-10-22:
Option to erase whole flash memory on target board is needed, in case someone loses their authentication password or just settings get corrupted. (occured at least once during workshops in Warsaw)

Marwe, 2019-06-30:
Also for passing on the device, to remove WiFi credentials.
This is implemented at least in GUI (version 0.2.2), so can be closed?

ricki-z, 2019-06-30:
Latest release version is 0.2.0, so this issue can't be closed. We should only close issues that are solved for ALL users, not only those that are able to compile the software.

Marwe, 2019-06-30:
Thanks @ricki-z, so we leave it open until the next release. Is there anything blocking the next release?
I guess the release is additionally listed, so anyone could use the current release 0.2, if necessary.

ricki-z, 2019-06-30:
@marwe It would be nice to have working build and release workflow at either travis or another build service. So we would be independent from a single user who has access to all needed platforms. At the moment the builds are done by Informatic and published at dropbox. Dropbox is the other reason not to link to the actual releases of Informatic. I tell users not to click to unknown Dropbox links as there are many infected files.

Informatic, 2019-06-30:
Just a minor note: https://dropbox.inf.re is just one of may subdomains hosted on my private server, no relation to dropbox. (just my random place to drop random files...)

Latest binary build is [published on github](https://github.com/opendata-stuttgart/airrohr-firmware-flasher/releases) but it'd be great to have it automatically handled by some sort of CI. I tried finishing Travis integration a long time ago, but failed for some reason, don't remember why.

ricki-z, 2019-06-30:
Hi Informatic, thanks for clarification about the host. But even this host may be too suspect for most users ;-)
The latest version published on Github is 0.2.0, but the source code is more actual and contains i.e. the "Erase Flash" option.

DeeKey, 2019-12-25:
When to expect builds of 0.3 version with FlashErase option?
Our local users ask for it. Some of them managed to passcode UI and now need to clean the flash easy way...

ricki-z, 2020-02-04:
Build of version 0.3.0 is available now.

DeeKey, 2020-07-07:
@Informatic - seems that this issue can be closed now.

closed 2020-09-02 by ricki-z