opendata-stuttgart/airrohr-firmware-flasher #3: add python installation to windows build script

ricki-z, 2018-10-24:
Installer options can be found at:
https://docs.python.org/3/using/windows.html#installing-without-ui

Informatic, 2018-10-24:
Alternatively we could just add python3 to cygwin dependencies. I'll test it in a minute if there would be any issues with that...

Informatic, 2018-10-24:
Oh well. It seems like pip in cygwin seems to have `sys.platform == 'cygwin'`, which breaks fetching of PyQt5. We'll just end up installing official python distribution. Too bad there doesn't seem to be a way to somehow install it in some local directory (like we already do with cygwin) without disrupting (user-)global python installation.

Informatic, 2018-10-24:
This has been fixed as of 5728442

closed 2018-10-24 by Informatic