opendata-stuttgart/airrohr-firmware-flasher #11: Please add empty image to the list of available firmware versions

mrgames13, 2019-07-07:
On some of my esp8266 devices occured errors with the WiFi connection. Until now, I always solved it by flashing an empty image on the device like in this tutorial by @ricki-z :

https://github.com/opendata-stuttgart/meta/wiki/Firmware-einspielen#l%C3%B6schen-der-konfiguration

It would be nice, if your tool could handle that in the future. The image file of the empty file system can be found here:

https://github.com/opendata-stuttgart/sensors-software/raw/master/airrohr-firmware/ppd42ns-wificonfig-ppd-sds-dht.spiffs.bin

Thanks for your work!

ricki-z, 2019-07-07:
The upcoming version of the flashing tool will have an "Erase flash" option.
We can't add this as a "normal" image as the config flash uses another start address.

ohheyitsdave, 2019-11-28:
Duplicate of #1 

:sparkles: In the coming version you will have the option to erase settings on the esp8266/NodeMCU

closed 2019-11-28 by ohheyitsdave