opendata-stuttgart/airrohr-firmware-flasher #12: Custom config file upload

Marwe, 2019-09-01:
Would it be feasible to create and upload a custom config to the SPIFF?
This way wifi and own API settings could be pre-defined.
Options for the GUI that come to my mind are with a checkbox or an additional button (upload custom config).
Editing a JSON (from a template) would be fine, I guess.

Informatic, 2019-09-01:
Generally speaking everythings possible. Issue with handling spiffs is that
python bindings use ctypes, which would require us to ship
platform-specific spiffs_.so binaries for all supported platforms. It's be
nice if someone sat down and did a basic pure-python reimplementation of
spiffs. I assume create-only implementation wouldn't be that hard, and
actually quite educational at that, but I've been very busy for the past
couple of weeks so can't commit to that.

Additional minor issue is that theoretically spiffs address offset might
change in the future, so we'd need to make some sanity check and verify if
there's some blank space for our fs (or there's some existing spiffs
binary) there.

ricki-z, 2019-09-01:
I wouldn't include this in the flashing tool.
This would be another place we need to change in case of new firmware functions. And at the moment there is only one person that can compile the tool for all needed platforms.
In case this person is leaving the project we can't add new features or change existing ones.
The debug mode will show the actually loaded json config. You can take this, create a new file "config.json" in the 'data' directory, make the needed changes and upload this file with the SPIFFS upload tool available for the Arduino IDE.

Marwe, 2019-09-03:
Ok, that sounds like it is already possible with external tools, which I was not aware of.
So no urgent need to integrate it into this tool, the above description/howto could be put in a more continuous place (somewhere in the wiki?).

closed 2019-09-03 by Marwe