opendata-stuttgart/airrohr-firmware-flasher #17: Add configuration of AP credentials during flashing

DeeKey, 2019-12-12:
Feature request:
Add possibility to flash AP username and password.
And maybe advanced mode to flash other information:
credentials for other data hosts (feinstaubapp, aircms and others)

If one can flash AP credentials in user mode then the configuration of a station will be easier. No need to run station in AP mode to enter AP credentials.

Marwe, 2020-01-06:
Without being too much into details, I guess, that writing a json config file to flash should do the trick? I think the esptool can directly write files there.
Then this could be done with a template json and a generic form creator?

ricki-z, 2021-05-12:
This should work with the actual sources (binaries will be published after some optimizations).