opendata-stuttgart/airrohr-firmware-flasher #20: Please add posibility to flash alpha build from remote URL or local file

DeeKey, 2020-01-03:
Sometimes I need  to flash alpha builds which are not available from dropdown.
I download them manually put in the folder and copy the name to the text box.

It is not a very convenient. Please add an option to flash from URL or from local file.

Informatic, 2020-01-03:
AFAIR, it should be possible to just copy over https://... link into version select box:
https://github.com/opendata-stuttgart/airrohr-firmware-flasher/blob/master/airrohr-flasher.py#L212-L223

DeeKey, 2020-01-08:
No it does not work like that (
Try yourself...

Informatic, 2020-01-08:
This has been implemented since the first published version for me, and has been tested/used on Linux systems. (and AFAIR worked on Windows as well) What platform are you using?

Example: https://d.inf.re/b/ffc776f1e74faa0ca4f50842e37ec567.webm - current master branch, disregard the HUAWEI Modem board, don't have esp8266 board at hand - file is still downloaded off my server, and should flash onto the board properly.

DeeKey, 2020-01-09:
This link does not work for me:
https://static.dmllr.de/airrohr/beta/builds-NRZ-2019-128-B8/latest_ru.bin

Always "Work in progress" message...

Informatic, 2020-01-25:
Current `master` branch seems to work for me on Linux with this link. `Work in progress` error shows up only if you press `Upload` button when the firmware is already being downloaded / flashed.

If error occurs during download, it should show up on the status bar (and next `Upload` button should again show `Downloading...`, or start flashing the corrupted binary, in case download got halted during download - this is a minor bug - after application restart download is attempted again)