opendata-stuttgart/airrohr-firmware-flasher #21: Segfault on Fedora 31

Qwertie-, 2020-01-07:
Attempted to download the prebuilt binary and run it but got this error:

```
$ '/home/user/Downloads/luftdaten-tool.linux-x64' 
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
/usr/lib/gio/modules/libdconfsettings.so: wrong ELF class: ELFCLASS32
Failed to load module: /usr/lib/gio/modules/libdconfsettings.so
GLib-GIO-Message: 21:05:40.956: Using the 'memory' GSettings backend.  Your settings will not be saved or shared with other applications.

(luftdaten-tool.linux-x64:8832): GdkPixbuf-WARNING **: 21:05:40.982: Cannot open pixbuf loader module file '/usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders.cache': No such file or directory

This likely means that your installation is broken.
Try running the command
  gdk-pixbuf-query-loaders > /usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders.cache
to make things work again for the time being.
Fontconfig warning: "/etc/fonts/fonts.conf", line 5: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/fonts.conf", line 6: unknown element "its:translateRule"
Fontconfig error: "/etc/fonts/fonts.conf", line 6: invalid attribute 'translate'
Fontconfig error: "/etc/fonts/fonts.conf", line 6: invalid attribute 'selector'
Fontconfig error: "/etc/fonts/fonts.conf", line 7: invalid attribute 'xmlns:its'
Fontconfig error: "/etc/fonts/fonts.conf", line 7: invalid attribute 'version'
Fontconfig warning: "/etc/fonts/fonts.conf", line 9: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/10-hinting-slight.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/10-hinting-slight.conf", line 5: unknown element "its:translateRule"
Fontconfig error: "/etc/fonts/conf.d/10-hinting-slight.conf", line 5: invalid attribute 'translate'
Fontconfig error: "/etc/fonts/conf.d/10-hinting-slight.conf", line 5: invalid attribute 'selector'
Fontconfig error: "/etc/fonts/conf.d/10-hinting-slight.conf", line 6: invalid attribute 'xmlns:its'
Fontconfig error: "/etc/fonts/conf.d/10-hinting-slight.conf", line 6: invalid attribute 'version'
Fontconfig warning: "/etc/fonts/conf.d/10-hinting-slight.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 5: unknown element "its:translateRule"
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 5: invalid attribute 'translate'
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 5: invalid attribute 'selector'
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 6: invalid attribute 'xmlns:its'
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 6: invalid attribute 'version'
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 5: unknown element "its:translateRule"
Fontconfig error: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 5: invalid attribute 'translate'
Fontconfig error: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 5: invalid attribute 'selector'
Fontconfig error: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 6: invalid attribute 'xmlns:its'
Fontconfig error: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 6: invalid attribute 'version'
Fontconfig warning: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/25-unhint-nonlatin.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/25-unhint-nonlatin.conf", line 5: unknown element "its:translateRule"
Fontconfig error: "/etc/fonts/conf.d/25-unhint-nonlatin.conf", line 5: invalid attribute 'translate'
Fontconfig error: "/etc/fonts/conf.d/25-unhint-nonlatin.conf", line 5: invalid attribute 'selector'
Fontconfig error: "/etc/fonts/conf.d/25-unhint-nonlatin.conf", line 6: invalid attribute 'xmlns:its'
Fontconfig error: "/etc/fonts/conf.d/25-unhint-nonlatin.conf", line 6: invalid attribute 'version'
Fontconfig warning: "/etc/fonts/conf.d/25-unhint-nonlatin.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/30-metric-aliases.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/30-metric-aliases.conf", line 5: unknown element "its:translateRule"
Fontconfig error: "/etc/fonts/conf.d/30-metric-aliases.conf", line 5: invalid attribute 'translate'
Fontconfig error: "/etc/fonts/conf.d/30-metric-aliases.conf", line 5: invalid attribute 'selector'
Fontconfig error: "/etc/fonts/conf.d/30-metric-aliases.conf", line 6: invalid attribute 'xmlns:its'
Fontconfig error: "/etc/fonts/conf.d/30-metric-aliases.conf", line 6: invalid attribute 'version'
Fontconfig warning: "/etc/fonts/conf.d/30-metric-aliases.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/40-nonlatin.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/40-nonlatin.conf", line 5: unknown element "its:translateRule"
Fontconfig error: "/etc/fonts/conf.d/40-nonlatin.conf", line 5: invalid attribute 'translate'
Fontconfig error: "/etc/fonts/conf.d/40-nonlatin.conf", line 5: invalid attribute 'selector'
Fontconfig error: "/etc/fonts/conf.d/40-nonlatin.conf", line 6: invalid attribute 'xmlns:its'
Fontconfig error: "/etc/fonts/conf.d/40-nonlatin.conf", line 6: invalid attribute 'version'
Fontconfig warning: "/etc/fonts/conf.d/40-nonlatin.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/45-generic.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/45-generic.conf", line 5: unknown element "its:translateRule"
Fontconfig error: "/etc/fonts/conf.d/45-generic.conf", line 5: invalid attribute 'translate'
Fontconfig error: "/etc/fonts/conf.d/45-generic.conf", line 5: invalid attribute 'selector'
Fontconfig error: "/etc/fonts/conf.d/45-generic.conf", line 6: invalid attribute 'xmlns:its'
Fontconfig error: "/etc/fonts/conf.d/45-generic.conf", line 6: invalid attribute 'version'
[... 53 lines cut ...]
Fontconfig error: "/etc/fonts/conf.d/65-nonlatin.conf", line 6: invalid attribute 'xmlns:its'
Fontconfig error: "/etc/fonts/conf.d/65-nonlatin.conf", line 6: invalid attribute 'version'
Fontconfig warning: "/etc/fonts/conf.d/65-nonlatin.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/69-unifont.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/69-unifont.conf", line 5: unknown element "its:translateRule"
Fontconfig error: "/etc/fonts/conf.d/69-unifont.conf", line 5: invalid attribute 'translate'
Fontconfig error: "/etc/fonts/conf.d/69-unifont.conf", line 5: invalid attribute 'selector'
Fontconfig error: "/etc/fonts/conf.d/69-unifont.conf", line 6: invalid attribute 'xmlns:its'
Fontconfig error: "/etc/fonts/conf.d/69-unifont.conf", line 6: invalid attribute 'version'
Fontconfig warning: "/etc/fonts/conf.d/80-delicious.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/80-delicious.conf", line 5: unknown element "its:translateRule"
Fontconfig error: "/etc/fonts/conf.d/80-delicious.conf", line 5: invalid attribute 'translate'
Fontconfig error: "/etc/fonts/conf.d/80-delicious.conf", line 5: invalid attribute 'selector'
Fontconfig error: "/etc/fonts/conf.d/80-delicious.conf", line 6: invalid attribute 'xmlns:its'
Fontconfig error: "/etc/fonts/conf.d/80-delicious.conf", line 6: invalid attribute 'version'
Fontconfig warning: "/etc/fonts/conf.d/90-synthetic.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/90-synthetic.conf", line 5: unknown element "its:translateRule"
Fontconfig error: "/etc/fonts/conf.d/90-synthetic.conf", line 5: invalid attribute 'translate'
Fontconfig error: "/etc/fonts/conf.d/90-synthetic.conf", line 5: invalid attribute 'selector'
Fontconfig error: "/etc/fonts/conf.d/90-synthetic.conf", line 6: invalid attribute 'xmlns:its'
Fontconfig error: "/etc/fonts/conf.d/90-synthetic.conf", line 6: invalid attribute 'version'
Fontconfig error: Cannot load config file from /etc/fonts/fonts.conf
Fontconfig warning: FcPattern object width does not accept value [70 100)
Segmentation fault (core dumped)
```

Qwertie-, 2020-01-07:
I just built the tool from source and it works fine.

tom-r, 2020-01-18:
I have the same issue on Fedora 31 with the delivered binary.
I think you should compile & link it with **static** libraries to include the dependencies.

1. It searches a path  /usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders.cache for that pixbuf thing... 
/usr/lib/x86_64-linux-gnu/ does not exist on Fedora31. The installation is in /usr/lib64/...
OK, you can set a link from /usr/lib to /usr/lib64 , but all those font.config errors/warnings remain and also the segmentation fault at the very end
So : the binary does not work under Fedora 31 !

2. Manual compilation :
the installation instructions in Readme.md for Fedora are crab
Duplicated, all in one line, dependencies missing,... 
Quote :
On Fedora: sudo dnf install **sudo dnf install** qt5-qttools-devel python3-devel **make ln -s /usr/bin/lrelease-qt5 /usr/bin/lrelease**
------------
At least packages sip and upx and gdk-pixbuf2 must be added.
I'm  still failing to compile on my own ...
Currently I'm struggling on the 'make deps dist', because of more missing dependency ...

ModuleNotFoundError: No module named 'PyQt5.sip'
make: *** [Makefile:16: gui/mainwindow.py] Fehler 1

Kind of a nightmare ...

Please fix.
 
BR,
Thomas

Informatic, 2020-01-25:
Indeed, sadly PyInstaller builds are a nightmare on Linux. I tried using staticx but failed miserably. Maybe migrating to appimage/snap or some other similar technology might be a good idea, as these start to get some wider adoption. Patches are welcome.

Regarding Fedora 31 - have you installed `python3-qt5` and/or `python3-qt5-sip` packages? (not sure how pyqt5 is split up on Fedora)

tom-r, 2020-01-25:
Hi,
I had installed python3-qt5 but this interfered with the installation via requirements.txt
there is no python3-qt5-sip, it's called python3-pyqt5-sip. And this is still installed.
So I uninstalled the official python3-qt5 package and tried to rebuild it with the _make deps_.
_make deps_ worked fine now without errors.
Even _make dist_ does finish but with warnings, and there I'm stuck (and I have no clue how to fix it). This is the output of make dist. Please note the sip-warning...
------------------------------
[tom@TR airrohr-firmware-flasher]$ make dist
python3 -m PyInstaller -y airrohr-flasher.spec
59 INFO: PyInstaller: 3.4
59 INFO: Python: 3.7.6
59 INFO: Platform: Linux-5.4.13-201.fc31.x86_64-x86_64-with-fedora-31-Thirty_One
67 INFO: UPX is available.
fatal: Kein Git-Repository (oder irgendein Elternverzeichnis bis zum Einhängepunkt /)
Stoppe bei Dateisystemgrenze (GIT_DISCOVERY_ACROSS_FILESYSTEM nicht gesetzt).
Can't extract git head, will use a dummy commit ID.
Command '['git', 'rev-parse', '--short', 'HEAD']' returned non-zero exit status 128.
87 INFO: Extending PYTHONPATH with paths
['/home/tom/git/airrohr-firmware-flasher',
 '/home/tom/git/airrohr-firmware-flasher']
87 INFO: checking Analysis
100 INFO: Building because /home/tom/git/airrohr-firmware-flasher/airrohrFlasher/_buildid.py changed
100 INFO: Initializing module dependency graph...
101 INFO: Initializing module graph hooks...
104 INFO: Analyzing base_library.zip ...
3439 INFO: Analyzing hidden import 'PyQt5.sip'
3468 INFO: running Analysis Analysis-00.toc
3493 INFO: Caching module hooks...
3502 INFO: Analyzing airrohr-flasher.py
3721 INFO: Processing pre-safe import module hook   urllib3.packages.six.moves
5026 INFO: Processing pre-safe import module hook   six.moves
7066 INFO: Loading module hooks...
7066 INFO: Loading module hook "hook-cryptography.py"...
7248 INFO: Loading module hook "hook-certifi.py"...
7250 INFO: Loading module hook "hook-PyQt5.QtGui.py"...
7437 INFO: Loading module hook "hook-PyQt5.QtWidgets.py"...
7581 INFO: Loading module hook "hook-xml.py"...
7657 INFO: Loading module hook "hook-encodings.py"...
7713 INFO: Loading module hook "hook-pydoc.py"...
7714 INFO: Loading module hook "hook-PyQt5.QtCore.py"...
7745 INFO: Loading module hook "hook-PyQt5.py"...
**7788 WARNING: Hidden import "sip" not found!**
7808 INFO: Looking for ctypes DLLs
**7916 WARNING: library Advapi32 required via ctypes not found
7949 WARNING: library setupapi required via ctypes not found**
7956 INFO: Analyzing run-time hooks ...
7965 INFO: Including run-time hook 'pyi_rth_qt5.py'
7974 INFO: Looking for dynamic libraries
11173 INFO: Looking for eggs
11174 INFO: Using Python library /usr/lib64/libpython3.7m.so.1.0
11185 INFO: Warnings written to /home/tom/git/airrohr-firmware-flasher/build/airrohr-flasher/warn-airrohr-flasher.txt
11243 INFO: Graph cross-reference written to /home/tom/git/airrohr-firmware-flasher/build/airrohr-flasher/xref-airrohr-flasher.html
11260 INFO: Appending 'datas' from .spec
11269 INFO: checking PYZ
11274 INFO: Building because toc changed
11274 INFO: Building PYZ (ZlibArchive) /home/tom/git/airrohr-firmware-flasher/build/airrohr-flasher/PYZ-00.pyz
11875 INFO: Building PYZ (ZlibArchive) /home/tom/git/airrohr-firmware-flasher/build/airrohr-flasher/PYZ-00.pyz completed successfully.
11885 INFO: checking PKG
11888 INFO: Building because toc changed
11888 INFO: Building PKG (CArchive) PKG-00.pkg
38015 INFO: Building PKG (CArchive) PKG-00.pkg completed successfully.
38029 INFO: Bootloader /usr/local/lib64/python3.7/site-packages/PyInstaller/bootloader/Linux-64bit/run
38029 INFO: checking EXE
38095 INFO: Building because toc changed
38095 INFO: Building EXE from EXE-00.toc
38096 INFO: Appending archive to ELF section in EXE /home/tom/git/airrohr-firmware-flasher/dist/airrohr-flasher
38312 INFO: Building EXE from EXE-00.toc completed successfully.
[tom@TR airrohr-firmware-flasher]$ 
-------------------

And when I finally try to execute _**make run**_, I get a segmentation fault ...
--------------------
[tom@TR airrohr-firmware-flasher]$ make run
python3 airrohr-flasher.py
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, webgl, xcb.

make: *** [Makefile:28: run] Aborted (Speicherauszug erstellt)
[tom@TR airrohr-firmware-flasher]$ 
--------------------

There is no xcb package I could install ...
Sorry I'm a python novice, all I know is, that it is "a programming language". I can't help you much.
It would be different story with C and C++ ...
Thanks for looking into it ...

BR,

Thomas

ricki-z, 2020-01-25:
@tom-r could you please try to reformat your comment? This is difficult to read at the moment.

tom-r, 2020-01-25:
done ...

ricki-z, 2020-02-04:
I hace checked the install instructions. They should now work with Fedora 31. Can you test this?
And there is an additionally download of the binary for Fedora 31.

tom-r, 2020-02-05:
Hi,
thanks for the quick fix. Here is an intermediate result :
the fedora31 binary starts w/o problems now on my system.
I don't have the nodemcu here at the moment, I will test to flash it on friday and come back with an answer.
I will also test & try to compile on friday according to your compile instructions.

BR,
Thomas

tom-r, 2020-02-07:
the given fedora 31 binary is working, I can flash the esp8266.
Compiling :
Still does not work.
I cannot use _mkvirtualenv_, not found...
_make deps_ installed 2 more packages, and reports no errors
_make dist_ still hat the same 3 warnings
_make run_ still crashes ...
I did a _make clean_ in between ..
 But it's not so important right now, I can use the binary.
Here comes the shell output :

[tom@TR airrohr-firmware-flasher]$ mkvirtualenv -p "$(which python3)" airrohr-firmware-flasher
bash: mkvirtualenv: Kommando nicht gefunden.

[tom@TR airrohr-firmware-flasher]$ make deps
python3 -m pip install --user -U -r requirements.txt
Looking in indexes: https://pypi.org/simple/
Requirement already up-to-date: altgraph==0.16.1 in /usr/local/lib/python3.7/site-packages (from -r requirements.txt (line 2)) (0.16.1)
Requirement already up-to-date: certifi==2018.8.24 in /usr/local/lib/python3.7/site-packages (from -r requirements.txt (line 3)) (2018.8.24)
Requirement already up-to-date: chardet==3.0.4 in /usr/lib/python3.7/site-packages (from -r requirements.txt (line 4)) (3.0.4)
Requirement already up-to-date: ecdsa==0.13.3 in /usr/local/lib/python3.7/site-packages (from -r requirements.txt (line 5)) (0.13.3)
Requirement already up-to-date: esptool==2.5.0 in /usr/local/lib/python3.7/site-packages (from -r requirements.txt (line 6)) (2.5.0)
Requirement already up-to-date: future==0.16.0 in /usr/local/lib/python3.7/site-packages (from -r requirements.txt (line 7)) (0.16.0)
Requirement already up-to-date: idna==2.7 in /usr/local/lib/python3.7/site-packages (from -r requirements.txt (line 8)) (2.7)
Requirement already up-to-date: macholib==1.11 in /usr/local/lib/python3.7/site-packages (from -r requirements.txt (line 9)) (1.11)
Requirement already up-to-date: netifaces==0.10.7 in /usr/local/lib64/python3.7/site-packages (from -r requirements.txt (line 10)) (0.10.7)
Requirement already up-to-date: pefile==2018.8.8 in /usr/local/lib/python3.7/site-packages (from -r requirements.txt (line 11)) (2018.8.8)
Requirement already up-to-date: pyaes==1.6.1 in /usr/local/lib/python3.7/site-packages (from -r requirements.txt (line 12)) (1.6.1)
Requirement already up-to-date: pyinstaller==3.6 in /home/tom/.local/lib/python3.7/site-packages (from -r requirements.txt (line 13)) (3.6)
Requirement already up-to-date: pyqt5-sip==4.19.19 in /home/tom/.local/lib/python3.7/site-packages (from -r requirements.txt (line 14)) (4.19.19)
Requirement already up-to-date: pyqt5==5.11.2 in /home/tom/.local/lib/python3.7/site-packages (from -r requirements.txt (line 15)) (5.11.2)
Requirement already up-to-date: pyserial==3.4 in /usr/local/lib/python3.7/site-packages (from -r requirements.txt (line 16)) (3.4)
Requirement already up-to-date: requests==2.20.0 in /usr/local/lib/python3.7/site-packages (from -r requirements.txt (line 17)) (2.20.0)
Requirement already up-to-date: urllib3==1.24.2 in /usr/local/lib/python3.7/site-packages (from -r requirements.txt (line 18)) (1.24.2)
Requirement already up-to-date: zeroconf==0.20.0 in /usr/local/lib/python3.7/site-packages (from -r requirements.txt (line 19)) (0.20.0)
Requirement already satisfied, skipping upgrade: setuptools in /usr/lib/python3.7/site-packages (from pyinstaller==3.6->-r requirements.txt (line 13)) (41.2.0)

[tom@TR airrohr-firmware-flasher]$ make dist
pyuic5 gui/mainwindow.ui -o gui/mainwindow.py
lrelease i18n/German.ts
Updating 'i18n/German.qm'...
    Generated 33 translation(s) (33 finished and 0 unfinished)
lrelease i18n/Polish.ts
Updating 'i18n/Polish.qm'...
    Generated 33 translation(s) (33 finished and 0 unfinished)
python3 -m PyInstaller -y airrohr-flasher.spec
34 INFO: PyInstaller: 3.6
34 INFO: Python: 3.7.6
35 INFO: Platform: Linux-5.4.15-200.fc31.x86_64-x86_64-with-fedora-31-Thirty_One
39 INFO: UPX is available.
fatal: Kein Git-Repository (oder irgendein Elternverzeichnis bis zum Einhängepunkt /)
Stoppe bei Dateisystemgrenze (GIT_DISCOVERY_ACROSS_FILESYSTEM nicht gesetzt).
Can't extract git head, will use a dummy commit ID.
Command '['git', 'rev-parse', '--short', 'HEAD']' returned non-zero exit status 128.
44 INFO: Extending PYTHONPATH with paths
['/home/tom/git/airrohr-firmware-flasher',
 '/home/tom/git/airrohr-firmware-flasher']
44 INFO: checking Analysis
52 INFO: Building because /home/tom/git/airrohr-firmware-flasher/airrohrFlasher/_buildid.py changed
52 INFO: Initializing module dependency graph...
54 INFO: Caching module graph hooks...
59 INFO: Analyzing base_library.zip ...
3260 INFO: Caching module dependency graph...
3352 INFO: running Analysis Analysis-00.toc
3376 INFO: Analyzing airrohr-flasher.py
3593 INFO: Processing pre-safe import module hook   urllib3.packages.six.moves
4800 INFO: Processing pre-safe import module hook   six.moves
6831 INFO: Analyzing hidden import 'PyQt5.sip'
6831 INFO: Processing module hooks...
6831 INFO: Loading module hook "hook-PyQt5.QtWidgets.py"...
[... 29 lines cut ...]
38982 INFO: Bootloader /home/tom/.local/lib/python3.7/site-packages/PyInstaller/bootloader/Linux-64bit/run
38982 INFO: checking EXE
38984 INFO: Building because toc changed
38985 INFO: Building EXE from EXE-00.toc
38992 INFO: Appending archive to ELF section in EXE /home/tom/git/airrohr-firmware-flasher/dist/airrohr-flasher
39145 INFO: Building EXE from EXE-00.toc completed successfully.

[tom@TR airrohr-firmware-flasher]$ make run
python3 airrohr-flasher.py
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
[... cut ...]
    from airrohrFlasher.qtvariant import QtGui, QtCore, QtWidgets
  File "/home/tom/.local/lib/python3.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 623, in exec_module
    exec(bytecode, module.__dict__)
  File "airrohrFlasher/qtvariant.py", line 4, in <module>
    from PyQt5 import QtGui, QtCore, QtWidgets
ImportError: /tmp/_MEIcEVt1S/libQt5Core.so.5: version `Qt_5_PRIVATE_API' not found (required by /tmp/_MEIcEVt1S/PyQt5/QtCore.so)
[31697] Failed to execute script airrohr-flasher
[tom@TR airrohr-firmware-flasher]$

ricki-z, 2020-02-07:
Please try this without a virtual environment.

rscircus, 2021-01-29:
+1 on Ubuntu 20.10.

ricki-z, 2021-02-06:
@rscircus what is the +1 for? For the problem or the solution?