opendata-stuttgart/airrohr-firmware-flasher #25: ESP32 NodeMCU CP2102 "chip_id is not supported for ESP32"

benjamq, 2020-01-22:
Hey guys,

i tried several times to flash my ESP32 CP2102, but the chip_id will not be written.
Is this board not supported or what am i doing wrong?

Thank you

Informatic, 2020-01-22:
This is likely. Support for ESP32 was not intended/required during
development, but this should be very easy to introduce.

benjamq, 2020-01-22:
I would appreciate it.

ricki-z, 2020-01-22:
If you look in our instructions you won't find any hint that the firmware is supporting the ESP32. This MCU isn't binary compatible to the ESP8266. So why did you think the firmware flashed with that tool would work with the ESP32? Just to get a hint where we need to inform the users about this.

benjamq, 2020-01-23:
I have copied the name "NodeMCU ESP8266, CPU/WLAN" from the construction manual into Amazon.
The first product looked good and i thought it would be fine.

> AZDelivery ESP32 NodeMCU Module WLAN WiFi Development Board mit CP2102 (Nachfolgermodell zum ESP8266) und gratis eBook!

benjamq, 2020-01-23:
I have bought the "right" one now.
For me its my first microcontroller, and my knowledge about the differences between ESP32/8266 and CP2102 is low. I thought CP2102 is the important number. And ESP32 is the follow-up of the 8266, so what will go wrong.

I found antoher tutorial:
https://www.teco.kit.edu/wp-content/grand-media/application/Anleitung_Crowdsensor_v085.pdf
they have a flasher for ESP32 and shares the data with you (Luftdaten)?!
So you have to circuit them befor my data reaches your server.

@ricki-z maybe my explanation helps you and the community

ricki-z, 2020-01-24:
You could try their firmware and flasher. With that the data should be send (also) to our servers.
Using their flashing tool to flash our firmware won't work. Our firmware binaries are only working on ESP8266.

closed 2020-02-01 by benjamq