opendata-stuttgart/airrohr-firmware-flasher #27: Sensiron SPS30 not found

atze09, 2020-03-01:
Hello guys,
First of all this is an awesome project.

I build 3x DNMS - T4+NodeMCU and 1x DNMS - T3.6+NodeMCU and have two new Sensiron SPS30.
Everything work fine, except the Feinstaub sensor Sensiron SPS30.
I tried all the 4 boards and two Sensors on each board and always get the followed error :
`13:23:04.564 -> Read SPS30...
13:23:04.564 -> Trying SPS30 sensor on 0x69H  ... not found
13:23:04.700 -> Check SPS30 wiring`

I measured the voltages of the sensor cable and they where at Pin 1=5V and Pin 2 and 3=3,3V.
Can anybody help me with this issue?
Maybe my sensors are defective..

Thank you
Danke.

atze09, 2020-03-02:
Works with latest Beta

closed 2020-03-02 by atze09

ricki-z, 2020-03-02:
Hello,
this is a firmware problem and not a problem of the flasher. So I move this issue to the sensor-software repo.