opendata-stuttgart/airrohr-firmware-flasher #28: "loading firmware list" does not work (firmare.sensor.community down?)

m-alt-f, 2020-03-21:
I'm using airRohr-firmware-flasher-0.3.0-Ubuntu_18.04_amd64 (and I have only used this platform).

About three weeks ago I built my 1st sensor and downloading the latest firmware worked like a charm.

Today, I tried to do the same. Two things went wrong.

1) This link: https://firmware.sensor.community/airrohr/flashing-tool/airRohr-firmware-flasher-0.3.0-Ubuntu_18.04_amd64 does not work anymore. I had to download the flashing tool from this GitHub.

2) The flasher tool I managed to download from this GitHub did not load the firmware list. The "Loading firmware list..." message kept showing up at the bottom.

I still managed to get the sensor up and running because of this video that the maintainer made: https://d.inf.re/b/ffc776f1e74faa0ca4f50842e37ec567.webm <-- doing it this way flashed the firmware beautifully and the sensor is now up and running.

m-alt-f, 2020-03-22:
It was obviously a temporary outage / server temporarily down or something like that. Just tested it - all works as it should again!

closed 2020-03-22 by m-alt-f

ricki-z, 2020-03-22:
At the moment there is sometimes too much traffic and some lines or servers may be overloaded. But we try to monitor this and to react to this.

Xennis, 2020-03-30:
https://firmware.sensor.community/airrohr/update/ seems to be down again. Hence the flash loader only shows a "Loading firmware list ..." and you can't use it for flashing :-(

I tried to change the `UPDATE_REPOSITORY` back to the old server `https://www.madavi.de/sensor/update/data/`. That's seem to work.