opendata-stuttgart/airrohr-firmware-flasher #32: failed config / IP unset

peter-scholtens, 2020-04-12:
I flashed the latest.bin (and also latest_nl.bin, latest_de.bin) as described here:
https://github.com/opendata-stuttgart/airrohr-firmware-flasher
But the webserver / Acces Point doesn't seem to start up. The log after pressing the reset button (I tried it several times):

```
airRohr: NRZ-2020-129/EN
mounting FS...
failed to open config file.
output debug text to displays...
Connecting to Freifunk-disabled
........................................
output debug text to displays...
Starting WiFiManager
AP ID: airRohr-XXXXXXXX
Password: 
scan for wifi networks...
AP Password is: 
Starting Webserver... (IP unset)
```
There is no Access Point / Wifi network visible. Also building with the sensors-software itself (using  Arduino IDE) and modifying ext_def.h didnot help. Did I need to set a config? I didn't found any description.

ricki-z, 2020-04-13:
The debug output seems okay so far. Could you please check that your 'client' has 2.4 GHz wifi enabled? The NodeMCU is only able to work with 2.4GHz wifi.

peter-scholtens, 2020-04-13:
I guess I had a faulty instance at hand. Tried another instance of the same board and it works as expected. So the issue can be closed. If you know a method to test the internals (RAM, flash?) of the faulty instance, let me know.

closed 2020-04-13 by peter-scholtens