opendata-stuttgart/airrohr-firmware-flasher #36: 0.3.1-MacOS.dmg is damaged and can't be opened

ddzieduch, 2020-06-12:
[image]

Previous releases work fine

MacOS Catalina
10.15.5 (19F101)

ricki-z, 2020-06-12:
I have checked both the version here at github.com and the version at firmware.sensor.community . Both are working as expected. Is it possible that your download wasn't finished correctly or that an anto-virus software is blocking the file?

ddzieduch, 2020-06-12:
I checked on a second Mac with the same version of the system and I have the same message.
And I also checked on the older one:

MacOS Snow Leopard
10.6.8 (10K549)

with the same result.

ricki-z, 2020-06-15:
Sorry Snow Leopard isn't supported anymore (Apple support ended in March 2016).

ricki-z, 2020-06-18:
It looks like a problem with the rights of downloaded files.
I can open the dmg file and "install" the app if I use the local dmg file. The same file downloaded from github or our own server doesn't work.
Seems to be a new "feature" of the actual MacOS version 😞

Tsvetkovv, 2021-02-01:
Changing attributes helped to me:
`xattr -cr /Applications/Sensor.Community\ Airrohr\ Flasher.app`

ddzieduch, 2021-02-03:
The above solution allows to run the application.