opendata-stuttgart/airrohr-firmware-flasher #37: Error msg after connection to ESP8266 was established

ordinatus1412, 2020-07-01:
I have an issue while flashing, the tool says "Timed out waiting for packet content" after the connection to the ESP8266 chip was established. The CH340 driver under Win10 is properly installed, the COM port automatically detected. Any idea what to do?

ricki-z, 2020-07-01:
Could you please post a picture of the module you try to flash?

ordinatus1412, 2020-07-02:
There u go

[image]

ricki-z, 2020-07-02:
Okay, this module should work.
You should use a short usb cable for flashing. And you could try to press and hold the "Flash" button just before you start the flashing. If the blue LED is blinking fast you can release the button.

michas79de, 2021-03-18:
Hi there!
It seems, I've the same problem.
I've bought this module: https://www.aliexpress.com/item/4001038369864.html
(The pics are right, the description is not, because it came with a CP2102 instead of the HC340.)
The cable is a 80 cm cable from my old phone. (Two 30 cm and a 70 cm cable powered up the module but didn't show anything connected.)
I'm trying with my PC and with my Laptop.

I've tried to upload immediately after connecting or later and also after pressing the flash button or the reset button (BTW: The flash button doesn't trigger any LED, the reset button triggers one single blue light up.)
Everytime I get the timeout error and have to physically dis- and reconnect to avoid the could-not-open-port-error for the next try.

Any hints?

[image]
[image]
[image]
[image]

ricki-z, 2021-03-18:
Are you sure that your cable can transfer data? Some times the USB cables are "power only".

michas79de, 2021-03-18:
Good question! I'm not sure about the three shorter ones. I've tried them only because of their length and didn't used them before.
But the other one can transfer data. I used it often to copy media between my Galaxy and my PC.
But I will also look for more cables in my house to test with them...

michas79de, 2021-03-18:
I've tried it also with Visual Studio Code without success.

The terminal log ends with

`[...]
Compiling .pio\build\nodemcuv2_en\FrameworkArduino\umm_malloc\umm_poison.c.o
Archiving .pio\build\nodemcuv2_en\libFrameworkArduino.a
Linking .pio\build\nodemcuv2_en\firmware.elf
Retrieving maximum program size .pio\build\nodemcuv2_en\firmware.elf
Checking size .pio\build\nodemcuv2_en\firmware.elf
Building .pio\build\nodemcuv2_en\firmware.bin
Creating BIN file ".pio\build\nodemcuv2_en\firmware.bin" using "C:\Users\Verwalter\.platformio\packages\framework-arduinoespressif8266\bootloaders\eboot\eboot.elf" and ".pio\build\nodemcuv2_en\firmware.elf"
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  39.4% (used 32296 bytes from 81920 bytes)
Flash: [=====     ]  53.6% (used 559772 bytes from 1044464 bytes)
after_build([".pio\build\nodemcuv2_en\firmware.bin"], [".pio\build\nodemcuv2_en\firmware.elf"])
Configuring upload protocol...
AVAILABLE: espota, esptool
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM3
Uploading .pio\build\nodemcuv2_en\firmware.bin
esptool.py v2.8
Serial port COM3
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 3c:61:05:d4:d7:cc
Uploading stub...
Running stub...
Stub running...
*** [upload] Error 2
Configuring flash size...

A fatal error occurred: Timed out waiting for packet header
================================================================================================== [FAILED] Took 24.70 seconds ==================================================================================================
Environment    Status    Duration
-------------  --------  ------------
nodemcuv2_en   FAILED    00:00:24.698
============================================================================================= 1 failed, 0 succeeded in 00:00:24.698 =============================================================================================The terminal process "C:\Users\Verwalter\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload', '--environment', 'nodemcuv2_en'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.`

That's why I think this is more of a hardware problem. Nevertheless, I would of course be happy about any hints. Or can the part also simply be scrap?!

michas79de, 2021-03-18:
It was my last sentence above - cheap scrap von china! 😤
I've re-soldered most connections. Especially the soldering between the two PCBs looked not great.

And now it worked instantly! Flasher started, module connected, upload, finished! 😃🥳

So maybe this can help you too @ordinatus1412