opendata-stuttgart/airrohr-firmware-flasher #45: Flashing the board only (without sensors). This should work?

eisengrau, 2020-11-23:
Hello,

As I understood, the software needs to be flased on the ESP8266 before the sensors are all assembled and mounted in their plance in the drainage pipe...

So, I tried to flash the firmware on the bare board itself, without attaching the other sensors (feinstaub, temperature, ...), and as I uderstood, by default, I should be presented with an SSID named Feinstaubsensor-(board ID number). Unfortunately this does not happen even if the flash was sucessful, according to the application. Is this normal, does the sensors attached need to initialize the firmware then I get the SSID, or there is something wrong with the flash/board?

ricki-z, 2020-11-23:
This should work as described. The first start may need some minutes as the filesystem is formatted. And some modules need a hard reset by a power cycle to initiate the wifi part correctly.

eisengrau, 2020-11-24:
Indeed, I guess the issue was on my side. I plugged in the usb cable to my PC directly, instead of an USB hub, and this time the image boot up properly and got the SSID visible.

ricki-z, 2020-11-27:
It's rather the missing 'hard reset' than the power supply. Can we (or you) close this issue?

eisengrau, 2020-11-27:
Yes, of course. Thank You!

closed 2020-11-27 by eisengrau