opendata-stuttgart/airrohr-firmware-flasher #46: VirusTotal detects 3/75

ThomasHeinrichSchmidt, 2020-11-24:
3 engines detected this file, see [Virus Total summary](https://www.virustotal.com/gui/file/8969bc62ec19ba599b6de34b676664627b7add81b5157fc0c725be7a53701f07/detection)

ricki-z, 2020-11-25:
These engines are detecting the flashing tool as "Potentially Unwanted Applications". Our flashing tool is selfextracting (contains the app itself and the scripts to flash the esp8266), compressed (to minimize time for download), needs to download a file from the internet (the latest firmware file) and tries to open a COM port. These behaviour is classified as "potentially unwanted". But it's exactly what the tool is made for. Good antivir software tries 
So temporary deactivating the category "Potentially Unwanted Applications (PUA)" should disable the message.
Better AV engines try to detect the URL that is opened and check the file that is downloaded (that's why most of the more common engines doesn't detect this file as false positive).

ThomasHeinrichSchmidt, 2020-11-25:
I see, all right then.

closed 2020-11-25 by ThomasHeinrichSchmidt