opendata-stuttgart/airrohr-firmware-flasher #50: Suddenly cannot connect to wifi (or maybe reboot?)

martinst1, 2021-01-03:
All of a sudden the sensor stopped working. I got it from the balcony, checked if all pins are properly connected, then rebooted and instead of connecting to the wifi network again the sensor created it's own network (airRohr.). When I enter http://192.168.4.1/config, type wifi credentials and press the restart button nothing happens, the airrohr network stays online and the sensor is not connected to the wifi network. Any idea what I am doing wrong and how to fix it? Sensor has been online for more than a year without any problems.

ricki-z, 2021-01-03:
Please check if there was any update or change in your wifi config. Sometimes router firmware updates are changing settings (i.e. in the last months some updates enabled WPA3 which may cause problems).

martinst1, 2021-01-03:
No changes what so ever. Also tried creating a hotspot from my phone - same behavior.

martinst1, 2021-01-03:
Also - the nova pm sensor blinks red in the back.

ricki-z, 2021-01-03:
If you connect to the sensor for configuration: Can you see your network (the one the sensor should be connected to) in the list of available networks?
Maybe there are some messages in the logs of your router. If so these could also be helpful.

martinst1, 2021-01-04:
Don't have access to the router. It's crazy, but it's my ISP policy.

Other than that yes - I can see the network and also other networks. I tried like I mentioned to connect to the generated by my phone access point with no results. Also tried to manually type the name of the network in the input field in case there are whitespaces or other obstacles by the auto copying but no.

botihu, 2021-01-04:
I would check for serial connection to the node controller is there any debug message printed or don't. Maybe trying to pump up debug level and do reboot the sensor.

Next step would be making a 'Factory' reset. If you can connect to the console via its own default AP (airohrxxxx SSID), you should find a 'Delete Configuration' menu option on the main page of the sensor.
Or it may be helpful to re-flash the sensor via the flasher tool. 

Hope any of those moving things forward and your sensor is coming back to alive.

martinst1, 2021-01-05:
Where are these debug messages stored and how do I access them? They are not in the interface at 192.168.4.1. 
I did try several times the reset button on the chip is there other way for factory reset? Also tried reflashing as per the installation steps 2 times, but nothing changes also.
Thanks for your suggestions, if you can provide me with step by step examples of how to do this things it will be cool.

botihu, 2021-01-05:
You may can try http://192.168.4.1/debug?lvl=5 (it may not works if sensor is not fully configured and running in setup mode.)

Debug messages can be found on serial console (using usb port of the controller and a terminal emultator like the popular minicom in linux or mac, or if you have windows, then putty also can do the serial terminal emulation). 

/** Actually I have used it with Linux, and not sure what driver need to install for windows to let terminal emulation to work. 
(used to I have USB-to-Serial [RS-232] converter but it had a Prolific PL2303 chip and its driver presented the COM port in windows.) **/ 

You can configure minicom with valid serial port settings. like /dev/ttyUSB0 (for linux) [or COM3 or COM5 (for windows - check in device manager which comm port) (or whatever device name recognized)] and any of standard serial communication parameters:
eg:
9600,8,n,1
115200,8,n,1 

Then if you plug in the pc you are running the minicom and resetting the controller you will see something in the terminal window. 
If it is just a garbage you need to chose a different port speed. 

Let me know if you need more detailed steps, I can grab back my sensor from my backyard for a few minutes to check the exact way how to get on the serial console.

elajkat, 2024-09-21:
Hi, sorry for piggybacking this issue, but I have similar with my airohr sensor and would like to connect to its serial to see the logs or any help what is wrong.
The issue is that it started to measure false data, or stopped measuring for hours, and now it is not visible on as client on my wifi routers list.
So back to serial console. I connected to my desktop machine's (ubuntu22.04) USB, and in dmesg I see these lines for it:
[cs szept 26 00:51:48 2024] usb 3-1: new full-speed USB device number 50 using xhci_hcd
[cs szept 26 00:51:48 2024] usb 3-1: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.54
[cs szept 26 00:51:48 2024] usb 3-1: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[cs szept 26 00:51:48 2024] usb 3-1: Product: USB2.0-Serial
[cs szept 26 00:51:48 2024] ch341 3-1:1.0: ch341-uart converter detected
[cs szept 26 00:51:48 2024] usb 3-1: ch341-uart converter now attached to ttyUSB0
[cs szept 26 00:51:49 2024] input: BRLTTY 6.4 Linux Screen Driver Keyboard as /devices/virtual/input/input67
[cs szept 26 00:51:49 2024] usb 3-1: usbfs: interface 0 claimed by ch341 while 'brltty' sets config #1
[cs szept 26 00:51:49 2024] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[cs szept 26 00:51:49 2024] ch341 3-1:1.0: device disconnected

And of course minicom just can't use and connect through ttyUSB0 as the device is really not there, could you tell me how can I go forward? Thanks in advance.

ricki-z, 2024-10-13:
This seems to be a local problem. BRLTTY and the usb3 driver both seem to try setting a config for this port.
Is there a chance to exclude ttyUSB0 from BRLTTY, i.e. with some kind of config setting?

elajkat, 2024-10-14:
Hi, thanks for answering. What do you mean by config setting? To tell the truth I am not familiar with the HW schematic of this device but if I understand well the issue happens after boot and I suppose that is also related to the issue with the wifi connection problem (an no measurements, not able to reach the web page and seemingly non-functioning device)

botihu, 2024-10-14:
Hi Lajos,

If you don't use Braille terminal at all, it is safe to completely remove
brlrtty from your linux host.

Chers,
   Boti