opendata-stuttgart/airrohr-firmware-flasher #51: Add ESP32 support

DeeKey, 2021-03-25:
We need to add support for ESP32 based board flashing in order to support Sensor Kit 2.

timmydan, 2021-10-06:
I had the same problem but got around it by finding and replacing both instances of "chip_id" (lines 496 and 642 in airrohr-flasher.py) with "read_mac" and then recompiling. This is new to me so I might not have understood all the complexities, but it seems that ESP32 doesn't use the command chip_id so read_mac is the closest thing. It ran fine after I made that change. I haven't yet connected it up to the other hardware, so consider this an answer-in-progress.

pjgueno, 2021-10-07:
It is in the esptool.py. We should write a conditional statement or use this function:
`
def chip_id(esp, args):
    try:
        chipid = esp.chip_id()
        print('Chip ID: 0x%08x' % chipid)
    except NotSupportedError:
        print('Warning: %s has no Chip ID. Reading MAC instead.' % esp.CHIP_NAME)
        read_mac(esp, args)
`