opendata-stuttgart/airrohr-firmware-flasher #53: Add 802.1X support (ex. Eduroam and authenticated network sessions)

vasconeves, 2021-04-10:
Hi!

I'm using about 50 feinstaub sensors but I have this recurring problem: school and universities have authenticated networks that require adding more information than just the network name and password. So I need to get creative and use a lot of workaround which are not really ideal in terms of security.

I'm wondering if this feature can be added in future updates.

Thanks.

Best regards,

Vasco

ricki-z, 2021-04-12:
This is a matter of available RAM. The existing implementations are very unstable. We tried to use 4096 bit certificates for HTTPS on our API servers, but had to go down to 2048 bit to get stable transmissions. For Eduroam Radius servers we can't tell the size of certificates used. So this may work with some systems but not all.

vasconeves, 2021-04-12:
Hello,

Ok...what about moving to esp32? Would that solve the problem?

Best,

V.

12/04/2021 à(s) 19:26: