opendata-stuttgart/airrohr-firmware-flasher #71: Not Working with sensor.community

liske310, 2024-03-13:
Today I see my data doesn`t send to sensor.community, but to others APIs (in example: Madavi, OpenSenseMap) works ok. I change from HTTPS to HTTP and now works, but can repair for works with https?

capitaenz, 2024-03-14:
My sensor 1398229 suddenly no longer sends data to the internet. Not even via Http. Is it due to the firmware NRZ-2020-133/DE (Nov 29 2020)?

DooMMasteR, 2024-04-08:
my sensor is now also not using some APIs anymore...

ricki-z, 2024-04-08:
There was an firmware update available since yesterday evening. Could you please check that this was loaded and installed?
Version is NRZ-2020-133. Maybe the device needs a power cycle after that.

liske310, 2024-04-12:
Via HTTP it`s works on firmware NRZ-2024-135/PL (Apr 9 2024) I check now via HTTPS - please take me a ~30minutes

liske310, 2024-04-12:
I confirm, after update works great :)